billw2/rpi-clone #156: Error: Could not stat device /dev/nvme0n1p

duvamduvam, 2023-06-18:
Hi,

From debian 12 I have always this error, with sd slot or usb it always top directly with this error.
Thanks for the help

dadou@5401:~/tmp/rpi-clone $ sudo ./rpi-clone mmcblk0
Error: Could not stat device /dev/nvme0n1p - Aucun fichier ou dossier de ce type.

dadou@5401:~/tmp/rpi-clone $ sudo ./rpi-clone sda1
Error: Could not stat device /dev/nvme0n1p - Aucun fichier ou dossier de ce type.

dadou@5401:~/tmp/rpi-clone $ ll
total 88
-rw-r--r-- 1 dadou dadou  1516 18 juin  18:27 LICENSE
-rw-r--r-- 1 dadou dadou 35460 18 juin  18:27 README.md
-rwxr-xr-x 1 dadou dadou 42807 18 juin  18:27 rpi-clone
-rwxr-xr-x 1 dadou dadou  2971 18 juin  18:27 rpi-clone-setup
dadou@5401:~/tmp/rpi-clone $ vim rpi-clone
dadou@5401:~/tmp/rpi-clone $ sudo ./rpi-clone sda1
Error: Could not stat device /dev/nvme0n1p - Aucun fichier ou dossier de ce type.

dadou@5401:~/tmp/rpi-clone $ ^C
dadou@5401:~/tmp/rpi-clone $ sudo sys-clone
Error: Could not stat device /dev/nvme0n1p - Aucun fichier ou dossier de ce type.

dadou@5401:~/tmp/rpi-clone $ sudo ./rpi-clone sda1
Error: Could not stat device /dev/nvme0n1p - Aucun fichier ou dossier de ce type.

dadou@5401:~/tmp/rpi-clone $ sudo ./rpi-clone sda
Error: Could not stat device /dev/nvme0n1p - Aucun fichier ou dossier de ce type.

framps, 2023-06-19:
Just a guess: It seems to me you use NVMe storage. That's not supported by rpi-clone as of now. Check https://github.com/billw2/rpi-clone/pull/147 which supports NVMe.

duvamduvam, 2023-06-20:
Hi thanks for the reply,
No I m using a regular sd card. Using my computer sd slot or usb adaptator give the same result.

framps, 2023-06-20:
Hm ... please show the result of `lsblk`

duvamduvam, 2023-06-21:
Yep :

NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0         7:0    0     4K  1 loop /snap/bare/5
loop1         7:1    0  55,7M  1 loop /snap/core18/2785
loop2         7:2    0  63,5M  1 loop /snap/core20/1891
loop3         7:3    0  63,4M  1 loop /snap/core20/1950
loop4         7:4    0  73,8M  1 loop /snap/core22/750
loop5         7:5    0  71,5M  1 loop /snap/cups/950
loop6         7:6    0 560,3M  1 loop /snap/freecad/726
loop7         7:7    0    55M  1 loop /snap/cups/872
loop8         7:8    0  91,7M  1 loop /snap/gtk-common-themes/1535
loop9         7:9    0 436,3M  1 loop /snap/kde-frameworks-5-96-qt-5-15-5-core20/7
loop10        7:10   0 437,3M  1 loop /snap/kde-frameworks-5-99-qt-5-15-7-core20/7
loop11        7:11   0 205,4M  1 loop /snap/rpi-imager/371
loop12        7:12   0  53,3M  1 loop /snap/snapd/19361
loop13        7:13   0  53,3M  1 loop /snap/snapd/19457
sda           8:0    1  59,5G  0 disk 
├─sda1        8:1    1   256M  0 part /media/dadou/bootfs
└─sda2        8:2    1  59,2G  0 part /media/dadou/rootfs
nvme0n1     259:0    0 931,5G  0 disk 
├─nvme0n1p1 259:1    0   100M  0 part /boot/efi
├─nvme0n1p2 259:2    0    16M  0 part 
├─nvme0n1p3 259:3    0 711,7G  0 part 
├─nvme0n1p4 259:4    0   639M  0 part 
├─nvme0n1p5 259:5    0  27,9G  0 part /
├─nvme0n1p6 259:6    0   977M  0 part [SWAP]
└─nvme0n1p7 259:7    0 190,1G  0 part /home

framps, 2023-06-21:
```
nvme0n1 259:0 0 931,5G 0 disk
├─nvme0n1p1 259:1 0 100M 0 part /boot/efi
├─nvme0n1p2 259:2 0 16M 0 part
├─nvme0n1p3 259:3 0 711,7G 0 part
├─nvme0n1p4 259:4 0 639M 0 part
├─nvme0n1p5 259:5 0 27,9G 0 part /
├─nvme0n1p6 259:6 0 977M 0 part [SWAP]
└─nvme0n1p7 259:7 0 190,1G 0 part /home
```
Here we are. You use an nvme drive :wink: . Try #147

duvamduvam, 2023-06-22:
I see ok, thanks for the help