billw2/rpi-clone #161: SDA partitions created then not found -> abort

Philtass, 2023-08-30:
I look at my save logs and I see that rpi-clone aborts with a message about sda1 and sda2 partitions not existing.
The proble seems to occur after partition resize. See file attached, please.
[rpi-clone verbose output.txt](https://github.com/billw2/rpi-clone/files/12478016/rpi-clone.verbose.output.txt)

Philtass, 2023-08-30:
PS: last successfull save was 19th january 2023

Philtass, 2023-08-30:
Thank's for fixing or tell me what to do for

Philtass, 2023-08-30:
Hi viniciuspaes !
sudo fdisk -l gives (at the end):
...
Disk /dev/sda: 29,8 GiB, 32010928128 bytes, 62521344 sectors
Disk model: Storage Device  
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Philtass, 2023-08-30:
Yes

Philtass, 2023-08-30:
previously I used "sudo rpi-clone sda -U" with same result

viniciuspaes, 2023-08-30:
It is weird that even parted is not recognizing SDA.

Double check if the command output the partitions:
cat /proc/partitions

Philtass, 2023-08-30:
$ cat /proc/partitions
major minor  #blocks  name

   1        0       4096 ram0
   1        1       4096 ram1
   1        2       4096 ram2
   1        3       4096 ram3
   1        4       4096 ram4
   1        5       4096 ram5
   1        6       4096 ram6
   1        7       4096 ram7
   1        8       4096 ram8
   1        9       4096 ram9
   1       10       4096 ram10
   1       11       4096 ram11
   1       12       4096 ram12
   1       13       4096 ram13
   1       14       4096 ram14
   1       15       4096 ram15
 179        0    7815168 mmcblk0
 179        1     262144 mmcblk0p1
 179        2    7544832 mmcblk0p2
   8        0   31260672 sda

Philtass, 2023-08-30:
Not for the moment. I'm not sure of parameters.

Philtass, 2023-08-30:
I see sda1 should be a fat32 and sda2 an ext3 but I never used command line to format a disk. I don't want to make a mess.

Philtass, 2023-08-30:
Coul'd you give the right commands ?

viniciuspaes, 2023-08-30:
Plug the external drive in your desktop computer. If using windows, try to format the drive and do some tests.

Then plug it again in the RPi and try the clone process. I did this way with my RPi 4 and worked in the first try.

Philtass, 2023-08-30:
I use linux. Have you a command line to format..

viniciuspaes, 2023-08-30:
Sure, hang on.

Philtass, 2023-08-30:
Ok try that

viniciuspaes, 2023-08-30:
To erase the external drive (double check if the external drive is mounted as /dev/sda)
sudo dd if=/dev/zero of=/dev/sda bs=16M

Using fdisk to create the partitions:
sudo fdisk /dev/sda

Philtass, 2023-08-30:
ok done that:
Device     Boot  Start      End  Sectors  Size Id Type
/dev/sda1         2048   526335   524288  256M  c W95 FAT32 (LBA)
/dev/sda2       526336 62924799 62398464 29,8G 83 Linux

viniciuspaes, 2023-08-30:
Nice, test the rpi-clone now.

Philtass, 2023-08-30:
I run rpi-clone, it seems to work (it's long)

Philtass, 2023-08-30:
bad news: 
sudo rpi-clone sda

Booted disk: mmcblk0 8.0GB                 Destination disk: sda 32.2GB
---------------------------------------------------------------------------
Part      Size    FS     Label           Part   Size    FS     Label
1 /boot   256.0M  fat32  --              1      256.0M  fat32  --
2 root      7.2G  ext4   rpidmtcz        2       29.8G  ext4   rpidmtcz
---------------------------------------------------------------------------
== SYNC mmcblk0 file systems to sda ==
/boot                 (47.5M used)   : SYNC to sda1 (256.0M size)
/                     (1.8G used)    : SYNC to sda2 (29.8G size)
---------------------------------------------------------------------------
Run setup script       : no.
Verbose mode           : no.
-----------------------:

Ok to proceed with the clone?  (yes/no): y

Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/sda2 on /mnt/clone
  => rsync // /mnt/clone with-root-excludes ...rsync: recv_generator: mkdir "/mnt/clone/var" failed: Bad message (74)
*** Skipping any contents from this failed directory ***
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1207) [sender=3.1.3]

mkdir: impossible de créer le répertoire « /mnt/clone/boot »: Message invalide
  Mounting /dev/sda1 on /mnt/clone/boot
mount: /mnt/clone/boot: cannot mount; probably corrupted filesystem on /dev/sda1.
    Mount failure of /dev/sda1 on /mnt/clone/boot.
  unmounting /mnt/clone
Aborting!

Philtass, 2023-08-30:
sorry for weird formated text

viniciuspaes, 2023-08-30:
The SD card you are trying to clone, what OS do you have installed?

It is possible to manually create the boot partition, and then you just rsync the second partition (sda2).

Philtass, 2023-08-30:
It's a standard raspbian, for a domoticz server.

Philtass, 2023-08-30:
on a RPI 3B

framps, 2023-08-30:
> I did'nt check logs before today, shame on me !

Why don't you create a small wrapper script which calls rpi-clone and sends a failure eMail to you if rpi-clone fails :wink:

Philtass, 2023-08-30:
Hi framps. Good idea !

Philtass, 2023-08-30:
But first I'd like to fix rpi-clone. Any Idea ?

viniciuspaes, 2023-08-30:
Do you have another external drive available? So we can isolate the problem.

Philtass, 2023-08-30:
I try with another SD card with only one partition. rpi-clone creates the 2 partitions and continues... for the moment

Philtass, 2023-08-30:
done successfully ! The problem came from the SD card I guess

Philtass, 2023-08-31:
Thanks for help and advices !

closed 2023-08-31 by Philtass