billw2/rpi-clone #163: sys-clone fails to mount cloned sd card on subsequent sync

timo12357, 2023-09-07:
I have an Orangepi 3LTS running Armbian 23.8.1 Bullseye with Linux 5.15.93-sunxi64.  Installed rpi-clone and renamed it to sys-clone by

`git clone https://github.com/matthijskooijman/rpi-clone.git
cd rpi-clone
git checkout support-fsuuid-and-armbian
sudo ./rpi-clone device`

sys-clone works perfect as long as I have the Armbian system on an SD card. Problems started when I moved the Armbian system to the Orangepi 3 LTS EMMC. The initial sync works, but sys-clone fails to do a subsequent sync complaining that it can not mount the clone disk:

$ sudo sys-clone sdc

Booted disk: mmcblk2 7,8GB                 Destination disk: sdc 31,3GB
---------------------------------------------------------------------------
Part     Size  FS  Label           Part   Size  FS    Label
1 root     7   2G  ext4            --    1        29  1G     ext4  --
---------------------------------------------------------------------------
== SYNC mmcblk2 file systems to sdc ==
/                     (2,9G used)    : SYNC to sdc1 (29,1G size)
---------------------------------------------------------------------------
Run setup script       : no.
Verbose mode           : no.
-----------------------:

Ok to proceed with the clone?  (yes/no): yes

Syncing file systems (can take a long time)
Syncing mounted partitions:
     e2label /dev/sdc1 --
  Mounting /dev/sdc1 on /mnt/clone
mount: /mnt/clone: mount(2) system call failed: Rakenne vaatii puhdistusta.
    Mount failure of /dev/sdc1 on /mnt/clone.
Aborting!`

The drive can be mounted manually after the initial sync with no errors. Running the subsequent sync with sys-clone makes the sd card unreadable.

matthijskooijman, 2023-09-07:
You're talking about `sys-clone`, but this repo contains only `rpi-clone`. Is that a typo?

AFAIU, you've cloned from SD-card to EMMC, which works, but then running from EMMC cloning back to SD card fails? Is that right?

You could try running `bash -x rpi-clone sdc 2>&1 | tee output.log`, that should print all the commands that are executed (and also log them to output.log). Maybe that provides a hint about what the script does that breaks the SD-card.

> The drive can be mounted manually after the initial sync with no errors. Running the subsequent sync with sys-clone makes the sd card unreadable.

You mean you can mount the SD card after the sync-to-EMMC (but before the sync-back-to-SD), but after the sync-back-to-SD fails, the SD-card is no longer mountable? That suggests to me that the script does something to the SD card that breaks the filesystem.

I'd also suggest you add some code formatting to your post (eg. use ` ``` `-delimited codeblocks) to make it a bit more readable.

timo12357, 2023-09-07:
> You're talking about `sys-clone`, but this repo contains only `rpi-clone`. Is that a typo?

I renamed rpi-clone to sys-clone as I am not using raspberry pi. It is the same code.

> AFAIU, you've cloned from SD-card to EMMC, which works, but then running from EMMC cloning back to SD card fails? Is that right?

I did not use rpi-clone to clone the system to EMMC, I used the Armbian native scripts to do it.

> You could try running `bash -x rpi-clone sdc 2>&1 | tee output.log`, that should print all the commands that are executed (and also log them to output.log). Maybe that provides a hint about what the script does that breaks the SD-card.

Will do and report

> You mean you can mount the SD card after the sync-to-EMMC (but before the sync-back-to-SD), but after the sync-back-to-SD fails, the SD-card is no longer mountable? That suggests to me that the script does something to the SD card that breaks the filesystem.

I can mount the SD card after the sync-**from**-EMMC-to-SD but before doing a subsequent sync from EMMC to SD.  If I repeat the sync from EMMC to SD the card is no longer mountable.

> I'd also suggest you add some code formatting to your post (eg. use `` ``` ``-delimited codeblocks) to make it a bit more readable.

Tried that but failed. Will try better.

timo12357, 2023-09-12:
This is what comes in the log:

[output.log](https://github.com/billw2/rpi-clone/files/12589322/output.log)

matthijskooijman, 2023-09-26:
I've looked at the log, but I'm puzzled why it does not work. It looks like the script mostly just collects info, and then the failing mount of /dev/sdb1 is pretty much one of the first things it does, so I do not see any preceding commands that look like they could kill the FS. The only thing that happens beforehand is setting the FS label, so maybe that breaks it?

Maybe you can verify this by trying it manually? e.g.:

 1. Do an initial sync from EMMC to SD
 2. Veryify mounting the SD-card works (on the device itself): `mount /dev/sdb1 /mnt/clone` and `umount /mnt/clone`
 3. Set the label: `e2label /dev/sdb1 --`
 4. Try mounting again: `mount /dev/sdb1 /mnt/clone`

timo12357, 2023-09-26:
Everything until
`e2label /dev/sdb1 --`
works. After that it fails:

`
$ sudo umount /mnt/clone
$ sudo e2label /dev/sdb1 --
$ sudo mount /dev/sdb1 /mnt/clone
mount: /mnt/clone: mount(2) system call failed: Rakenne vaatii puhdistusta.
`
The error text direct translation is "Structure needs cleaning"

timo12357, 2023-09-29:
If I run rsync without the e2label command it works just fine:

`$ sudo rsync -acxHAX --info=progress2 / /mnt/clone --exclude='/mnt/clone/*'
        256,621   0%   16.28kB/s    0:00:15 (xfr#12, to-chk=0/81770)`

What is the e2label command good for?

matthijskooijman, 2023-10-01:
> What is the e2label command good for?

It sets the filesystem label. In this case there is none (or it is literally "--", not sure). That should be fairly harmless command, though it *does* modify the filesystem, and in this case, for some reason, that gets messed up.

Maybe you could try running `sudo e2label /dev/sdb1 somelabel` instead and see if that also breaks? If not, the `--` label is somehow problematic...

timo12357, 2023-10-01:
Unfortunately same error:

`$ sudo e2label /dev/sdb1 somelabel`
`$ sudo mount /dev/sdb1 /mnt/clone`
`mount: /mnt/clone: mount(2) system call failed: Rakenne vaatii puhdistusta.`

Is the e2label command necessary in the context of rpi-clone? Why is it used here?

matthijskooijman, 2023-10-01:
It is used to copy the filesystem label from the source filesystem to the target filesystem, to provide a sync that is as complete as possible. Also, the filesystem label is sometimes used for mounting the filesystem (check `/etc/fstab` and `/proc/cmdline`), though using partition or filesystem IDs is often used instead.

So I guess you could remove the e2label command and see if that works (would be a good debug step as well - maybe e2label is not the problem, but just exposes a different problem..)

timo12357, 2023-10-01:
`change_label()
        {
        pnum=$1
        fs_type=$2
        dev=$3

        dst_part_label "$pnum" "$fs_type" label
        if [ "$label" != "" ] && [[ "$fs_type" == *"ext"* ]]
        then
                echo "     e2label $dev $label"
#               e2label $dev $label
        elif [ "$label" != "" ] && [[ "$fs_type" == *"fat"* ]]
        then
                echo "     fatlabel $dev $label"
                fatlabel $dev $label
        fi
        }

`