billw2/rpi-clone #167: Problem on Orange Pi 3B, Debian Bookworm cloning from /dev/mmcblk0 to /dev/mvme0n1

b-morgan, 2023-10-07:
I have an Orange Pi 3B with an eMMC (256GB) and an NVMe (256GB) on which I have installed Debian 12 (Bookworm). I have installed rpi-clone 2.0.22 and copied the files to /local/user/sbin as suggested in the README.md.

Here is a lsblk:
```
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mtdblock0     31:0    0    16M  0 disk
mmcblk0      179:0    0   233G  0 disk
├─mmcblk0p1  179:1    0     1G  0 part /boot
└─mmcblk0p2  179:2    0 229.6G  0 part /var/log.hdd
                                       /
mmcblk0boot0 179:32   0     4M  1 disk
mmcblk0boot1 179:64   0     4M  1 disk
zram0        254:0    0   1.9G  0 disk [SWAP]
zram1        254:1    0    50M  0 disk /var/log
nvme0n1      259:0    0 238.5G  0 disk
├─nvme0n1p1  259:1    0     1G  0 part
└─nvme0n1p2  259:4    0 231.9G  0 part
```
When I attempt to do the clone, I get a Warning that requires a response but I have no idea where it comes from or what response will continue.

```
root@orangepi3b:~# sys-clone /dev/nvme0n1 -f

  Target disk nvme0n1 ends with a digit so may be a partition.
  sys-clone requires disk names like 'sda' and not partition names like 'sda1'.

Continue anyway?  (yes/no): y
Warning: Not all of the space available to /dev/nvme0n1 appears to be used, you can fix the GPT to use all of the space
(an extra 11563696 blocks) or continue with the current setting?
```
Here is the output from fdisk for both the source and the target devices:
```
root@orangepi3b:~# fdisk /dev/mmcblk0

Welcome to fdisk (util-linux 2.38.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

This disk is currently in use - repartitioning is probably a bad idea.
It's recommended to umount all file systems, and swapoff all swap
partitions on this disk.

Command (m for help): p

Disk /dev/mmcblk0: 232.96 GiB, 250139901952 bytes, 488554496 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: C1C93DD0-252E-1A49-8268-4B0FB4B1A113

Device           Start       End   Sectors   Size Type
/dev/mmcblk0p1   61440   2158591   2097152     1G Linux extended boot
/dev/mmcblk0p2 2158592 483655679 481497088 229.6G Linux filesystem

Command (m for help): q

root@orangepi3b:~# fdisk /dev/nvme0n1

Welcome to fdisk (util-linux 2.38.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): p
Disk /dev/nvme0n1: 238.47 GiB, 256060514304 bytes, 500118192 sectors
Disk model: KBG40ZNS256G NVMe KIOXIA 256GB
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: C1C93DD0-252E-1A49-8268-4B0FB4B1A113

Device           Start       End   Sectors   Size Type
/dev/nvme0n1p1   61440   2158591   2097152     1G Linux extended boot
/dev/nvme0n1p2 2158592 488552447 486393856 231.9G Linux filesystem
```
Can you tell me what response is needed to continue (and where this warning comes from)?

b-morgan, 2023-10-07:
Using `bash -x sys-clone /dev/nvme0n1 2>&1 | tee output.log` I was able to discover that the message is generated by parted and the valid responses are `Fix/Ignore`. This allowed the script to continue but there is a new error:
```
  Mounting /dev/nvme0n12 on /mnt/clone
+ mount /dev/nvme0n12 /mnt/clone
mount: /mnt/clone: special device /dev/nvme0n12 does not exist.
       dmesg(1) may have more information after failed mount system call.
+ echo '    Mount failure of /dev/nvme0n12 on /mnt/clone.'
    Mount failure of /dev/nvme0n12 on /mnt/clone.
+ '[' '' '!=' '' ']'
+ echo 'Aborting!'
Aborting!
+ exit 1
```
From the previous fdisk output the partitions are `nvme0n1p1` and `nvme0n1p2`.

framps, 2023-10-07:
You should check #147 because rpi-clone does not support NVMe

b-morgan, 2023-10-07:
@framps Thanks! That issue gave me the clue but there were two lines that needed to be changed, 576:
```
	if [[ $disk == *"mmcblk"* || $disk == *"nvme"* ]]
```
and 1062:
```
	if [[ $dst_disk == *"mmcblk"* || $disk == *"nvme"* ]]
```

STR4NG3RdotSH, 2024-04-21:
> @framps Thanks! That issue gave me the clue but there were two lines that needed to be changed, 576:
> 
> ```
> 	if [[ $disk == *"mmcblk"* || $disk == *"nvme"* ]]
> ```
> 
> and 1062:
> 
> ```
> 	if [[ $dst_disk == *"mmcblk"* || $disk == *"nvme"* ]]
> ```
Great tweak, @b-morgan 
This got my attempt to clone SD -> NVMe past the issue I was having, after fixing your typo. 
Line 1062 should actually read:
`if [[ $dst_disk == *"mmcblk"* || $dst_disk == *"nvme"* ]]`
Just a heads up for anyone else who arrives here and doesn't notice it.
Cloning process completed and reported success.

STR4NG3RdotSH, 2024-04-21:
Disregard my responses. The best option is to use Jeff's (@geerlingguy) version/fork:
https://rpi-clone.jeffgeerling.com

After troubleshooting above for a short time I full removed and reinstalled his fork, ran clone to my nvme with no flags or script edits. Immediate success and system booted great.

framps, 2024-04-21:
Yes, everybody should head over to Jeff's repo. Bills repo is no longer maintained and does not work on Bookworm any more.