billw2/rpi-clone #170: RPi5

scargill, 2023-10-28:
Not an issue, Bill - just enquiring (but couldn't think of anywhere else to post this)..  Will RPi-Clone continue to function in the RPi5 without modification?

framps, 2023-10-29:
I think that's not dependent on the new HW but the SW required by RPi5. As far as I know RPi5 requires bookworm and rpi-clone works with bookworm if you apply the patch listed in https://github.com/billw2/rpi-clone/issues/168.

Melluish, 2024-01-01:
I successfully clones an RPi5 SD card last night to another SD card connected via a USB adaptor. The only issue was that for some reason the PARTUUID didn't get updated in `cmdline.txt` so had to be manually changed afterwards for it to boot.

framps, 2024-01-01:
@Melluish See #168 how to fix your PARTUUID update issue