billw2/rpi-clone #172: Q: clone to USB Flash Stick for backup; how to restore?

chrissv, 2023-11-12:
My apologies if I missed this in the documentation, but I couldn't find it.

In my implementation, instead of using an SD card in a USB-to-SD adapter as the clone destination, I have a regular USB flash stick (just because I have a bunch around). rpi-clone has no problems cloning to the USB flash device.

So what happens when/if I have an SD card failure, and I need to use/restore this cloned USB flash drive to a new SD card? Is there a tool to copy/restore from the USB stick back to an SD card (presumably on another machine, since the RPi isn't running)? Does anything (PARTUUID) need to be fixed up afterwards on the SD card?

framps, 2023-12-30:
I don't know for sure - but as far as understand just use rpi-clone to clone the USB stick to a new SD card. Just try it with a new SD card :wink:

chrissv, 2024-01-23:
> I don't know for sure - but as far as understand just use rpi-clone to clone the USB stick to a new SD card. Just try it with a new SD card 😉

Do you know the command for rpi-clone to use a different source than the booted device? (the "-m" option adds an additional mounted directory to sync, but the root is aways synced)

framps, 2024-01-23:
Unfortunately not.