billw2/rpi-clone #174: Unable to boot after cloning from SD to SSD

Rainz3, 2024-01-04:
I've gone through the install and after running `rpi-clone sda` it does it's transfers and looks like it's successful but it doesn't boot. It gives an error

`Gave up waiting for root file system device. Common problems:
 - Boot args (cat /proc/cmdline)
     - Check rootdelay= (did the system wait long enough?)
 - Missing modules (cat /proc/modules; ls /dev)
 ALERT! PARTUUID= d2a8e92e-02 does not exist. Dropping to shell!`

I've tried two different usb drives. A Centon and a kingston a400. Both give the same error. I also now can't boot off the original SD card as it gives the same error as well.

framps, 2024-01-04:
Please check #168

Rainz3, 2024-01-04:
> Please check #168

Ok, I was able to get one of my pi4s working using the fork but the other is still giving the error. I deleted the rpi-clone directory and redid the steps using https://github.com/framps/rpi-clone.git. It goes through like normal but I get the same error when I restart the pi using just the usb drive. I'm not sure if there is a more thorough way of removing the previous rpi-clone.

framps, 2024-01-04:
> but the other is still giving the error.

rpi-clone misses a debug log :cry: 

If the backup doesn't boot usually it's because UUIDs were not updated for whatever reason. Try [syncUUIDs](https://github.com/framps/raspberryTools/blob/master/syncUUIDs.sh) which will check whether the UUIDs are OK. If not use option `-u` to fix the incorrect UUIds.

Rainz3, 2024-01-04:
I've tried the -u and that ran through but same results. I tried the script but I'm not sure if I am running it write. I moved it over to the pi and ran it using ./syncUUIDs.sh. It gives me 'missing device to update UUIDs.'. I tried taking the drive a doing a fresh reformat on my windows machine and rerunning it, same results though now I'm back to not being able to boot at all with the SD card.

framps, 2024-01-04:
> I'm back to not being able to boot at all with the SD card.

That's bad but given the fact there is no debug log available in rpi-clone there is no way to help you - at least from my side.