billw2/rpi-clone #176: A friendly fork of rpi-clone (updated for modern Pi)

geerlingguy, 2024-02-07:
Just wanted to post not only thanks to @billw2 for creating and maintaining this lovely tool for years, but also to announce a friendly fork of the project: https://rpi-clone.jeffgeerling.com

Using many contributions from @framps (and help from others), it modernizes this project to work on the latest versions of Raspberry Pi OS, and supports features like cloning to and from PCIe NVMe SSDs.

The forked codebase is here: https://github.com/geerlingguy/rpi-clone

seamusdemora, 2024-02-08:
Cool... look forward to trying it soon.

barisahmet, 2024-02-20:
I was having issues clone my sdcard to nvme drive on my pi5 with the original script. Tried @geerlingguy's fork worked like a charm. Thanks both!

geerlingguy, 2024-02-20:
@seamusdemora re: '... but what is a "freindly" fork?'

Basically, a fork of an open source project that is done in consideration for the original repository, and doesn't mean to take away from the original spirit/community/authorship of the project. See: https://github.blog/2022-04-25-the-friend-zone-friendly-forks-101/

The alternative is people get angry, try to scrub @billw2's name from the repository and code, etc. In my case, I'm happy with the legacy of this codebase and will mainly serve to keep it updated for modern Pi OS and Raspberry Pi features (like NVMe support).

Lightwel, 2025-08-20:
Thanks to @geerlingguy for this. 

I want to use it to clone first two partitions of my RPi5 NVME to SSD to act as a bootable recovery disk in case I stuff the rootfs partition installing BVM (thanks for the video on that as well BTW). 

$ sudo rpi-clone -f2 sda seemed to work but on checking /mnt/clone/boot I get cmdline.txt in /boot instead of in boot/firmware as expected.

Additionally, in mnt/clone there is another /mnt/clone which is however empty.

Am I doing something wrong?

Have not tried booting from the new sda pending an answer to this and also the following question which I think ought to be clear to me but isn't:

in order to get a bootable recovery disk do I have to change the partuuid's on the sda clone, and in order to boot the sda clone and then restore to the original NVME do I have to change the partuuid's again after restoration?  The original NVME includes a data partition as well as the two RPi partitions. I can backup the data partition separately, if that is a simpler solution.

Tks Lightwell