billw2/rpi-clone #178: My rPi SD Card that I attempted to Clone won't boot anymore

ndehchef, 2024-03-04:
I have an rPi 4
Revision	: c03115
Serial		: 10000000a414a5d1
Model		: Raspberry Pi 4 Model B Rev 1.5

I was running 64 bit bookworm
I attempted to follow the instructions here:
https://gist.github.com/avoidik/d8fc39a372db680090edd5322d60848f 
Got as far as step 3 where I ran just sudo rpi-clone -s
I didn't continue because I wanted to sort some issues with my USB
After I shutdown, I attempted to start up my rPi but it wouldn't boot anymore. 
Any help would be appreciate.. I have the SD card now plugged in on the USB port and I can see all my files are still there. I am running my rPi an another SD card that I had created. still bit  bookworm.. just not with all my critical files on it.

ndehchef, 2024-03-04:
Thank you very much. I will. in the meantime, is there anything I can do to get my sd card to boot?

framps, 2024-03-04:
Your clone doesn't boot because the PARTUUID in /boot/firmware/cmdline.txt was not updated because of change in Bookworm. Either update the PARTUUID with vi or use my tiny synctool https://github.com/framps/raspberryTools/blob/master/syncUUIDs.sh

framps, 2024-03-04:
@ndehchef Just detected syncUUIDs will not work any more because in a second step they removed the link from /boot/cmdline.txt to /boot/firmware/cmdline.txt :cry:.

So you have to update /boot/firmware/cmdline.txt manually or wait until I fixed this in the script.

framps, 2024-03-04:
@ndehchef Forget my previous comment I just deleted. Everything is fine with syncUUIDs. I was puzzled because one leftover statement in the code.

ndehchef, 2024-03-04:
Thanks.. But I didn't clone. Its my original SD that's not booting. I never went through with the cloning and stopped before running the actual clone command. The only command I ran after cloning the repo and doing the setup is sudo  rpi-clone -s

framps, 2024-03-04:
My fault. I thought you're talking about the clone. 

I did a quick code check and as far as I can see rpi-clone-setup just updates /etc/hostname and /etc/hosts. But this should happen on a device mounted on /mnt/clone - not on the running OS :thinking: .

But the system should boot even with another hostname. Nevertheless I would check these two files and correct the hostname if it was changed. 

Do you have a screen connected and see any messages during boot? 

If you don't get it fixed I suggest to head over to [another clone and maintained repository](https://github.com/geerlingguy/rpi-clone/issues) and create an issue there :wink:

ndehchef, 2024-03-05:
I googled and found that something was messed up with the partitions and a script usb-boot help fix it. Sorry it has taken over a day to resolve all the issues. Thanks for your help. I will head over to the new repo and start again with the cloning.

closed 2024-03-05 by ndehchef