billw2/rpi-clone #180: Cloning an MVMe M2 SSD connected to a PCIe NVMe card - Could not stat device /dev/nvme0n1p

jaballarin, 2024-05-04:
When cloning an MVMe M2 SSD connected to a PCIe NVMe card that is the boot drive of the Raspberry OS, the following error occurs:
    Error: Could not stat device /dev/nvme0n1p - No such file or directory.
I have tried various command syntaxes, but it always gives that error.

Thank you.

framps, 2024-05-04:
This repository is no longer maintained. Please head over to [this](https://github.com/geerlingguy/rpi-clone/) repository and make sure you use the latest code from there.

jaballarin, 2024-05-04:
> This repository is no longer maintained. Please head over to [this](https://github.com/geerlingguy/rpi-clone/) repository and make sure you use the latest code from there.

Thanks a lot!

I will do so.
Best regards.