billw2/rpi-clone #182: cmdline.txt not update on cloned disk

gtnscebba, 2024-09-24:
I installed version 2.0.22 on a Raspberry PI4b.
The Linux version is SMP PREEMPT Debian 1:6.6.47-1+rpt1 (2024-09-02) aarch64 GNU/Linux.

Cloning is successful, but at the end it only updates fstab and not cmdline.txt.

The result is that the cloned SD is not able to boot correctly.

By manually updating cmdline.txt with the correct PARTUUID, the SD boots correctly. I can see the correct PARTUUID from sudo blkid.

closed 2024-09-24 by gtnscebba

mitoyosh, 2026-08-18:
I hit exactly this on a Pi 4B — cloned the running SD card to a USB SSD, `fstab` on the destination was correctly rewritten with the new PARTUUID, but `cmdline.txt` still pointed at the *source* disk's PARTUUID. If you flip the boot order without catching it, you either silently fall back to booting the SD card or hit a kernel panic waiting for a root device that isn't there — and nothing in the clone process warns you either way.

Wrote up the full diagnosis and the exact commands to check for it before touching boot order: https://homelabpostmortem.com/2026/08/16/rpi-clone-partuuid-trap/ (also packaged a small dry-run script if that's useful).

framps, 2026-08-18:
@mitoyosh Thank you very much for your detailed explanation of the issue and the fix. But this repo is no longer maintained for a long time. There exists a [maintained fork](https://github.com/geerlingguy/rpi-clone/) with right now 1003 additional commits on top to this repo, the issue is fixed for a long time already and there is no landmine in latest rpi-clone ;-) 

Please update your pages and links accordingly.

Given the UUID mismatch happens quite often I wrote some time ago a tool called [syncUUIDs](https://github.com/framps/raspberryTools/blob/master/syncUUIDs.sh)  which checks and optionally updates the UUIDs in fstab and commandline if they don't match the devices UUIDs.

mitoyosh, 2026-08-18:
Thanks — you're right on both counts, and the correction turned out to be more interesting than the original writeup.

I checked before updating: `billw2/rpi-clone` last pushed 2024-03-21, `geerlingguy/rpi-clone` 2026-04-19, and the fork does handle this. I'd also mischaracterised the mechanism. I'd said rpi-clone doesn't touch `cmdline.txt`; it does — the version here hardcodes `${clone}/boot/cmdline.txt`, which since Bookworm is the placeholder the OS leaves behind:

```
DO NOT EDIT THIS FILE

The file you are looking for has moved to /boot/firmware/cmdline.txt
```

So the fix-up runs, greps a file with no PARTUUIDs in it, matches nothing, and reports success — while the real `/boot/firmware/cmdline.txt` keeps the source disk's PARTUUID. Zero matches is indistinguishable from "nothing needed changing", which is why it's silent. The fork's runtime `cmdlinedir` detection avoids the whole thing.

Post is updated: corrected mechanism, a dated correction notice, the fork as the recommended tool, and a pointer to `syncUUIDs` (which is a nicer answer than my one-off script for anyone hitting this after the fact). Appreciate you taking the time.