billw2/rpi-clone #185: Unable to boot into original SD card after cloning from SD card to SSD

GuiFlam, 2025-04-03:
Hi, 

I successfully cloned my system from my SD card to an external SSD. I thought it went well. Now when I tried to boot from the ssd, it didn't work I got some errors about not finding the sd card so then when I tried to boot with the original SD card that was previously working, I now get this error when I boot from the original SD card.

```
You are in emergency mode. After logging in, type "journalctl -xb" to view system logs, "systemctl reboot" to reboot, "systemctl default" or "exit" to boot into the default mode.

Cannot open access to console, the root account is locked.
See sulogin(8) man page for more details.

Press Enter to continue.
```

After I click on enter, I see:

```
Reloading syste, manager configuration
Starting default.target
```
And about 30sec later, I get back the `You are in emergency mode` message.

The only command I ran was sudo rpi-clone sda to clone my system, and now I'm unable to boot from the sd card I cloned from. 

Can someone help?

Thanks

framps, 2025-04-03:
This repo is no longer maintained. Please visit https://github.com/geerlingguy/rpi-clone/ and use the latest code from there.

GuiFlam, 2025-04-03:
Well it's a little too late I already ran the code I'm  stuck now it would be great if it was written that this repo is no longer maintained..

framps, 2025-04-03:
Only the owner of this repo can update the README and mark this repo as no longer maintained - unfortunately 😩