billw2/rpi-clone #190: Is possible to exclude specific mount ?

araminimichael, 2025-08-30:
Hello, I don't think I found the doc for that. Is it possible to exclude cases? Because I have folders in /mnt/samba. And in mnt and the output of the script gives me:
```
Syncing file systems (can take a long time)
Syncing mounted partitions:
  Mounting /dev/sda1 on /mnt/clone
  => rsync // /mnt/clone with-root-excludes ...rsync: [sender] readlink_stat("/mnt/remote-media") failed: Stale file handle (116)
rsync: [sender] readlink_stat("/mnt/smb/addons") failed: Stale file handle (116)rsync: [sender] readlink_stat("/mnt/smb/share") failed: Stale file handle (116)
rsync: [sender] readlink_stat("/mnt/smb/media") failed: Stale file handle (116)
rsync: [sender] readlink_stat("/mnt/smb/ssl") failed: Stale file handle (116)
rsync: [sender] readlink_stat("/mnt/smb/backup") failed: Stale file handle (116)rsync: [sender] readlink_stat("/mnt/smb/config") failed: Stale file handle (116)IO error encountered -- skipping file deletion
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
===============================
Done with clone to /dev/sda
   Start - 20:35:54    End - 20:39:54    Elapsed Time - 4:00
Cloned partitions are mounted on /mnt/clone for inspection or customizing.
Hit Enter when ready to unmount the /dev/sda partitions ...
  unmounting /mnt/clone
===============================
```

framps, 2025-08-30:
Please check `--exclude`option.

araminimichael, 2025-08-31:
> Please check `--exclude`option.

Hi, Thank you for your answer. I see in fact "{--exclude=PATTERN}" But I don't see any other explanation. I'm quite a beginner. 

I just have to add --exclude=/mnt/samba/* or /mnt/*? 
Or do I have to insert one after the other? 

Thank you.

araminimichael, 2025-09-01:
For newbie like me.

See https://chatgpt.com/share/68b550fe-7e64-800c-9671-bde78798eae6